wokwi/rp2040js #4: MULS assembler code

chegewara, 2021-04-14:
Hi,
like i mentioned in chat there is bug in MULS assembler code, thats why its issue with tests, but actual code is correct.
Creating opcode takes registers in reverted order:
```
export function opcodeREV(Rd: number, Rn: number) {
  return (0b1011101000 << 6) | ((Rn & 0x7) << 3) | (Rd & 0x7);
}
```
and for MULS there is bug:
```
export function opcodeMULS(Rn: number, Rdm: number) {
  return (0b0100001101 << 6) | ((Rn & 7) << 3) | (Rdm & 7);
}
```

https://github.com/wokwi/rp2040js/blob/master/src/utils/assembler.ts#L130

urish, 2021-04-14:
Thanks for pointing this out!

If I understand correctly, the datasheet says that Rn comes first?

```
MULS{<q>} {<Rd>,} <Rn>, <Rm>
```

Obviously, there's something that I'm missing here, but I can't really tell what?

chegewara, 2021-04-14:
It should be like this:
```
export function opcodeMULS(Rdm: number, Rn: number) {
  return (0b0100001101 << 6) | ((Rn & 7) << 3) | (Rdm & 7);
}
```

urish, 2021-04-17:
Thanks!

If you feel like sending a pull request that will be awesome!

closed 2021-05-17 by chegewara