wokwi/rp2040js #25: npm start gives error: "Error: ENOENT: no such file or directory, open 'hello_uart.hex'"

anurag3301, 2021-05-14:
I am getting this error when I do npm start.
`
> rp2040js@0.3.6 start
> ts-node demo/emulator-run.ts

Error: ENOENT: no such file or directory, open 'hello_uart.hex'
    at Object.openSync (fs.js:458:3)
    at Object.readFileSync (fs.js:360:35)
    at Object.<anonymous> (/home/anurag/try/rp2040js/demo/emulator-run.ts:9:16)
    at Module._compile (internal/modules/cjs/loader.js:1138:30)
    at Module.m._compile (/home/anurag/try/rp2040js/node_modules/ts-node/src/index.ts:1056:23)
    at Module._extensions..js (internal/modules/cjs/loader.js:1158:10)
    at Object.require.extensions.<computed> [as .ts] (/home/anurag/try/rp2040js/node_modules/ts-node/src/index.ts:1059:12)
    at Module.load (internal/modules/cjs/loader.js:986:32)
    at Function.Module._load (internal/modules/cjs/loader.js:879:14)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:71:12)
`

anurag3301, 2021-05-14:
I also tried grepping the file, but nothing found
command: `ls -lRa rp2040js | grep hello_uart.hex`

urish, 2021-05-14:
Hi, thanks for reporting!

hello_uart.hex is not part of the project, but you can build it from the [pi pico examples repo](https://github.com/raspberrypi/pico-examples/tree/master/uart/hello_uart).

I'm not sure if this is a good idea to include it in this repo or not, but in any case, here's my compiled version in case someone wants it:

[hello_uart.hex.txt](https://github.com/wokwi/rp2040js/files/6478897/hello_uart.hex.txt)

(I added .txt to the filename, as GitHub doesn't allow uploading .hex files. So just rename it back to hello_uart.hex)

anurag3301, 2021-05-14:
@urish Thanks buddy, finally got it working.

closed 2021-05-14 by anurag3301