wokwi/rp2040js #29: sbcs error on xPSR

Turro75, 2021-05-16:
another bug catched by gdbdiff.
This time I left enabled the interrupts forcing the emulator following the path of the real pico. Tested on test-printf.ino

[diff.txt](https://github.com/wokwi/rp2040js/files/6489051/diff.txt)
[testdiff.txt](https://github.com/wokwi/rp2040js/files/6489052/testdiff.txt)

Mismatch after  3587  compared instructions

Register	StartValue	Emulator	Silicone
r0  		0x00000000	0x00000000	0x00000000
r1  		0x2000d1a0	0x2000d1a0	0x2000d1a0
r2  		0x00000009	0x00000009	0x00000009
r3  		0xffffffff	0xffffffff	0xffffffff
r4  		0x2000cf34	0x2000cf34	0x2000cf34
r5  		0x2000d1a0	0x2000d1a0	0x2000d1a0
r6  		0x2000ce30	0x2000ce30	0x2000ce30
r7  		0x10011648	0x10011648	0x10011648
r8  		0x20002c58	0x20002c58	0x20002c58
r9  		0x00000001	0x00000001	0x00000001
r10  		0x10011648	0x10011648	0x10011648
r11  		0x00000001	0x00000001	0x00000001
r12  		0x2000267d	0x2000267d	0x2000267d
sp  		0x2000cc80	0x2000cc80	0x2000cc80
lr  		0x10008c9f	0x10008c9f	0x10008c9f
pc  		0x10008022	0x10008024	0x10008024
xPSR* 		0x81000000	0x81000000	0x41000000
Flags		 [N---] 	 [N---] 	 [-Z--]

0x10008022 <+6>:	98 41	sbcs	r0, r3

Turro75, 2021-05-16:
Uri,
You are getting some noise from my repo, sorry for that, still learning the github way

closed 2021-05-16 by urish

urish, 2021-05-16:
Released this bugfix as 0.3.9

PR #30 (2021-05-16): fix(rp2040): Fix SBCS flags handling
Close #29