wokwi/rp2040js #35: hello_timer wrong printf output

Turro75, 2021-05-19:
running hello_timer.hex from pico examples result in this output:

> Unimplemented peripheral PADS_BANK0_BASE read from 4
> Unimplemented peripheral PADS_BANK0_BASE write to 4: -129
> Unimplemented peripheral IO_BANK0_BASE write to 4: 2
> Unimplemented peripheral PADS_BANK0_BASE read from 8
> Unimplemented peripheral PADS_BANK0_BASE read from 8
> Unimplemented peripheral PADS_BANK0_BASE write to 8: -129
> Unimplemented peripheral IO_BANK0_BASE write to c: 2
> UART sent:  H
> UART sent:  e
> UART sent:  l
> UART sent:  l
> UART sent:  o
> UART sent:   
> UART sent:  T
> UART sent:  i
> UART sent:  m
> UART sent:  e
> UART sent:  r
> UART sent:  !
> UART sent:  
> UART sent:  
> 
> SEV
> UART sent:  T
> UART sent:  i
> UART sent:  m
> UART sent:  e
> UART sent:  r
> UART sent:   
> UART sent:  V
> UART sent:   
> UART sent:  f
> UART sent:  i
> UART sent:  r
> UART sent:  e
> UART sent:  d
> UART sent:  !
> UART sent:  
> UART sent:  
> 
> SEV
> UART sent:  R
> UART sent:  e
> UART sent:  p
> UART sent:  e
> UART sent:  a
> UART sent:  t
> UART sent:   
> UART sent:  a
> UART sent:  t
> UART sent:   
> UART sent:  V
> UART sent:  V
> UART sent:  V
> UART sent:  V
> UART sent:  V
> UART sent:  V
> UART sent:  V
> UART sent:  V
> UART sent:  V
> UART sent:  V
> UART sent:  V
> UART sent:  V
> UART sent:  V
> UART sent:  V
> UART sent:  V
> UART sent:  V
> UART sent:  V
[... 545 lines cut ...]
Repeat at 8504053                                                                   
Repeat at 9004053                                                                   
Repeat at 9504053                                                                   
Repeat at 10004053                                                                  
cancelled... 1                                                                      
Done      
```                                                                          

Investigating with gdbdiff it looks like at some point the silicone runs code from bootrom and the emulator gets mad.
As far as I've seen it seems it does not properly handle the 16bit instructions when the memory address is in the bootrom memory.

on the silicone set a breakpoint at 0x10000416  (line 45 of hello_timer.c) then after roughly 70ish si it goes to bootrom memory area.

I tried this test, it actually use LSLS instruction instead of BL because not able to read/write properly at bootrom addresses:
```
it('should execute `bl 0x23c2` instruction', async () => {
    await cpu.setPC(0x00000122);
    await cpu.writeUint32(0x00000122, opcodeBL(0x23c2));
    await cpu.singleStep();
    const registers = await cpu.readRegisters();
    expect(registers.pc).toEqual(0x000024e8);
    expect(registers.lr).toEqual(0x00000127);
  });

```

urish, 2021-05-19:
Hmm... so the emulator doesn't print the numbers correctly?

I suspect it may have to do with the integer divider that we still haven't implemented. 

Can you check if there are any writes to SIO addresses at 0xd0000060 or 0xd0000068 ?

Turro75, 2021-05-19:
[out.txt](https://github.com/wokwi/rp2040js/files/6511777/out.txt)

correct, a lot of missing sio warnings, I didn't see anything to 68 but 60 there are, a lot.
also seen that __wrap_vprintf goes to bootrom

urish, 2021-05-20:
Thanks for attaching the logs!

These are SIO reads (and not writes) but I can see some reads come from the integer divider area:

```
Read from invalid SIO address: 60
Read from invalid SIO address: 64
Read from invalid SIO address: 74
Read from invalid SIO address: 70
```

70 is DIV_QUOTIENT,  
74 is DIV_REMAINDER

I guess `printf()` is trying to divide the number by 10 each time to get one digit at a time, and it uses the integer divider for that. But we haven't implemented it yet, so it spits out garbage...

One workaround may be to disable the hardware divider. There's probably a way to tell them compiler not to use the hardware divider, I'll try to google that / look into the SDK sources

urish, 2021-05-20:
I think this is the code is the build system which enables the hardware divider:

https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_divider/CMakeLists.txt

So maybe adding 

```
pico_set_divider_implementation(hello_timer compiler)
```

to the project's CMakeLists.txt will do the trick?

Turro75, 2021-05-20:
As it seemed not so complicated (for sure i introduced bugs, no doubt on this) I implemented the hardware divider and , as usual, You were right: the printf output is now correct

urish, 2021-05-20:
You implemented the hardware divider? Cool! I'll check the pull request soon.

closed 2021-05-24 by urish