wokwi/rp2040js #46: wrong subs register overflow flag

Turro75, 2021-05-28:
catched by gdbdiff:
Register	StartValue	Emulator	Silicone
r0  		0x853d1eee	0x053d1eee	0x053d1eee
r1  		0x00020000	0x00020000	0x00020000
r2  		0xfffe0000	0xfffe0000	0xfffe0000
r3  		0x80000000	0x80000000	0x80000000
r4  		0x853d1eee	0x853d1eee	0x853d1eee
r5  		0xea718d61	0xea718d61	0xea718d61
r6  		0x00000002	0x00000002	0x00000002
r7  		0x00000001	0x00000001	0x00000001
r8  		0xffffffff	0xffffffff	0xffffffff
r9  		0x0000000d	0x0000000d	0x0000000d
r10  		0x0a7a3dc7	0x0a7a3dc7	0x0a7a3dc7
r11  		0x20000b60	0x20000b60	0x20000b60
r12  		0x2000017d	0x2000017d	0x2000017d
sp  		0x20041f90	0x20041f90	0x20041f90
lr  		0x10001a81	0x10001a81	0x10001a81
pc  		0x200001cc	0x200001ce	0x200001ce
xPSR* 		0x81000000	0x31000000	0x21000000
Flags		 [N---] 	 [--CO] 	 [--C-]

0x200001cc <__udivsi3+80>:	c0 1a	subs	r0, r0, r3

urish, 2021-05-28:
Good catch! Thanks :)

I think we also have the same bug in `CMP (register)` and `CMP (register) encoding T2`, and maybe also in CMN? not sure about that one.

Mind checking?

Turro75, 2021-05-30:
found another test case that fails on both encodings

it('should set flags Z C when executing `cmp r3, r11` instruction when r3=0 and r11=0x80000000', async () => {
    await cpu.setPC(0x20000000);
    await cpu.writeUint16(0x20000000, opcodeCMPregT2(r3, r11));
    await cpu.setRegisters({ r3: 0x0, r11: 0x80000000 });
    await cpu.singleStep();
    const registers = await cpu.readRegisters();
    expect(registers.N).toEqual(false);
    expect(registers.Z).toEqual(true);
    expect(registers.C).toEqual(true);
    expect(registers.V).toEqual(false);
  });

  it('should set flags N V when executing `cmp r3, r7` instruction when r3=0 and r7=0x80000000', async () => {
    await cpu.setPC(0x20000000);
    await cpu.writeUint16(0x20000000, opcodeCMPregT1(r3, r7));
    await cpu.setRegisters({ r3: 0x0, r7: 0x80000000 });
    await cpu.singleStep();
    const registers = await cpu.readRegisters();
    expect(registers.N).toEqual(true);
    expect(registers.Z).toEqual(false);
    expect(registers.C).toEqual(false);
    expect(registers.V).toEqual(true);
  });

urish, 2021-05-30:
> ```ts
> it('should set flags Z C when executing cmp r3, r11 instruction when r3=0 and r11=0x80000000', async () => {
>   await cpu.setPC(0x20000000);
>   await cpu.writeUint16(0x20000000, opcodeCMPregT2(r3, r11));
>   await cpu.setRegisters({ r3: 0x0, r11: 0x80000000 });
>   await cpu.singleStep();
>   const registers = await cpu.readRegisters();
>   expect(registers.N).toEqual(false);
>   expect(registers.Z).toEqual(true);
>   expect(registers.C).toEqual(true);
>   expect(registers.V).toEqual(false);
> });

Strange, I don't see why the Z flag is set. The datasheet says:

```
shifted = Shift(R[m], shift_t, shift_n, APSR.C);`
(result, carry, overflow) = AddWithCarry(R[n], NOT(shifted), ‘1’);`
APSR.Z = IsZeroBit(result);
```

so, if we put the values of the registers, we get:

```
result = 0 + ~0x80000000 + 1;
```

which is 0x80000000, and obviously not 0... So I'm probably missing something?

urish, 2021-05-30:
Oh, I see what's the problem. `cmp r3, r11` is not a valid thumb instruction. You can do `cmp r11, r3`, but not the other way around. So when you write `cmp r3, r11`, it actually encoded `cmp r3, r3`. Which explains why the zero flag is set...

urish, 2021-05-30:
But the second issue (with the overflow bug) seems like a real issue, similar to the SUBS issue

Turro75, 2021-05-30:
> Oh, I see what's the problem. `cmp r3, r11` is not a valid thumb instruction. You can do `cmp r11, r3`, but not the other way around. So when you write `opcodeCMPregT2('cmp r3, r11')`, it actually encodes `cmp r3, r3`. Which explains why the zero flag is set...

```
<Rn> and <Rm> not both from R0-R7
if n < 8 && m < 8 then UNPREDICTABLE;
```
I though it meant at least one register > r7

urish, 2021-05-30:
> I though it meant at least one register > r7

Right, I missed that. I actually see that the bug is in `opcodeCMPregT2`. It should be `Rm & 0xf` instead of `Rm & 0x7`:

```diff
 export function opcodeCMPregT2(Rn: number, Rm: number) {
-   return (0b01000101 << 8) | (((Rn >> 3) & 0x1) << 7) | ((Rm & 0x7) << 3) | (Rn & 0x7);
+   return (0b01000101 << 8) | (((Rn >> 3) & 0x1) << 7) | ((Rm & 0xf) << 3) | (Rn & 0x7);
 }
```

Turro75, 2021-05-30:
ok I'm fixing it

commit b00fb760 (2021-05-30): fix(rp2040): CMP T1 & T2  flags

also fixes a bug in opcodeCMPregT2 in assembler.ts

fix various missing await in instructions test

prettier refactor of some parts on rp2040.ts

close #46
closed 2021-05-30 by urish