wokwi/rp2040js #77: MicroPython ticks from epoch is wrong

pyTony, 2021-08-09:
I did in micropython:

>>> import time
>>> time.time()
4790818803

Then in my PC I converted it to datetime:

>>> import datetime
>>> print(datetime.datetime.fromtimestamp(4790818803)
2121-10-25 10:00:03   

In my PC now:
 >>> time.time()
1628497438.6996312
>>> datetime.datetime.fromtimestamp(1628497438.6996312)                                                                 datetime.datetime(2021, 8, 9, 11, 23, 58, 699631)

urish, 2021-08-09:
Thanks! Is this behavior different from the physical Pi Pico? (I don't have one at the moment to test it on)

pyTony, 2021-08-09:
Unfortunately I have not Pico myself.

Best,
Tony

pyTony, 2021-08-09:
I googled micropython, and it looks epoch is different actually.
Time Epoch: Unix port uses standard for POSIX systems epoch of 1970-01-01 00:00:00 UTC. However, embedded ports use epoch of 2000-01-01 00:00:00 UTC.

Time from the number was in future, with later epoch one would expect that. So maybe this is just micropython difference from standard. Interesting was that it was whole number, but PC number had fractions of second.

pyTony, 2021-08-09:
I did calculation in Spreadsheet

Epoch	Now	Days	Seconds
1.1.2000	9.8.2021	7891	681782400
1.1.1970	9.8.2021	18848	1628467200

So expected value would be around 681782400. But maybe it is not expected to be RTC in many embedded applications.

PR #110 (2022-12-06): Fix #77
Fix #77 
```
# Emulator
>>> import time
>>> time.time()
1609459200
# PC
>> import datetime
>> datetime.datetime.fromtimestamp(1609459200)
datetime.datetime(2021, 1, 1, 8, 0)
```

Work as expected, the date is set by MicroPython
https://github.com/micropython/micropython/blob/1b774b373eb7fafebaa43adf9e086a690461201d/ports/rp2/main.c#L105-L115

But currently, there is no clock to update its time. Always get the same value.
closed 2022-12-06 by urish