wokwi/rp2040js #79: Latest Stable MicroPython (20210902) does not provide REPL at startup

DevyHeavy, 2021-09-25:
Trying to use the 20210902 MicroPython .uf2 with the newest 0.14.2 rp2040js build does not start/display the friendly REPL

urish, 2021-09-25:
Did older versions work for you?

DevyHeavy, 2021-09-25:
Yes

Using the 20210618 MicroPython .uf2 works fine

urish, 2021-09-25:
Can you please try to find the earliest version where it stops working?

DevyHeavy, 2021-09-25:
Stable MicroPython builds for the rp2040 jump from 20210618 to 20210902 (https://micropython.org/download/rp2-pico/) and so the earliest version it stops working is technically 20210902

urish, 2021-09-25:
Got it, there seem to be a lot of changes between these two versions, so it might take me some time to pinpoint the issue. Thanks for reporting!

What's your use case for the MicroPython simulation?

DevyHeavy, 2021-09-25:
Turning mcu.logger to 'Debug' level in 'demo/micropython-run-ts' displays a lot of USB transfer messages with working 20210618 but none in 20210902.

My use case is simply messing around with it. I may try to use it to simulate graphics on a webpage using a MicroPython ssd1306 screen library in the future. Although, I have not seen an example of importing a 'custom' python file/module

urish, 2021-09-28:
Should be fixed now!

commit c17c452b (2021-09-28): fix(usb): broken on MicroPython 1.17 #79

fix #79
closed 2021-09-28 by urish