wokwi/rp2040js #85: PIO `mov x, pins` input mask not applied

martin2250, 2021-10-30:
The following code produces different results on wokwi.com and actual hardware. Tested on MicroPython7.0.0.

When testing on hardware, only the first `in_pin_count` pins appear on the output. On wokwi.com, all pins appear on the output, regardless of `in_pin_count`.

I think the following line should get masked with ((1 << in_pin_count) - 1): https://github.com/wokwi/rp2040js/blob/235119548d16efe4818ed5122a5574dc0a6f5894/src/peripherals/pio.ts#L267

Cheers,
Martin

```python
import array
import board
import rp2pio
import adafruit_pioasm

code = """
  set x, 5
  mov x, pins
  in x, 32
  push block
"""

assembled_code = adafruit_pioasm.assemble(code)

sm2 = rp2pio.StateMachine(
    assembled_code,
    frequency=2000,
    in_pin_count = 8,
    first_in_pin=board.GP0,
    push_threshold=32,
    auto_push=False,
)

while True:
  buf = array.array('L', [0])
  sm2.readinto(buf)
  print(','.join(hex(x) for x in buf))
```

urish, 2021-10-30:
Thanks for reporting! I guess you refer to CircuitPython and not MicroPython?

urish, 2021-10-30:
Managed to reproduce: https://wokwi.com/arduino/projects/313883083894948417

urish, 2021-10-30:
Further investigation shows the hardware does not support `in_pin_count`. It's internal to CircuitPython, and is not passed down to the PIO hardware:

https://github.com/adafruit/circuitpython/blob/8657c6a2f5d55c29906fd95736fce640fea46fcb/ports/raspberrypi/common-hal/rp2pio/StateMachine.c#L151-L322

So unfortunately the proposed fix won't do the trick. Can you please elaborate on what you are observing on the real hardware v.s. in the simulator?

martin2250, 2021-10-31:
Ok, that's really odd... I just checked the C-SDK as well as the datasheet and there really is no mention of the number of input pins...

I double-checked it on hardware again, it definitely behaves differently depending on the value of `in_pin_count`, as you would expect. I used a low impedance (aka a straight wire) to connect the 3.3V pin to different GPIOs and only the first `in_pin_count` pins set the appropriate bit on the output.

Maybe it has something to do with the configuration of the GPIO pins?

urish, 2021-10-31:
Probably so. Though, if you used a low impedance connection, it means that either the input pad has to be disconnected from the PIO peripheral, or the bits should be masked out somewhere.

Perhaps this piece of code could be relevant:

https://github.com/adafruit/circuitpython/blob/8657c6a2f5d55c29906fd95736fce640fea46fcb/ports/raspberrypi/common-hal/rp2pio/StateMachine.c#L484-L488

I'm yet to decipher what it actually does...

urish, 2021-10-31:
Looking at the data sheet - the input pins are always connected to the PIO peripheral, regardless of the selected function:

[image]

Can you please check what the GPIO7_STATUS status reads when you have in_pin_count = 8 v.s. in_pin_count = 3?

specifically, the following two bits seem interesting:

[image]

martin2250, 2021-10-31:
Just came home, I didn't find a way to access raw registers on Circuitpython (I only used CPy because micropython isn't offered by wokwi.com).

Also tried to reproduce it using the C-SDK, which resulted in the same behavior as the simulation. You might be onto something there, GPIOX_CTRL.INOVER might be the solution to the problem. I'll look around in the CPy source code, maybe they disable all input pins by default.

martin2250, 2021-10-31:
No mention of INOVER anywhere... The C-SDK doesn't mention it in any function that doesn't have 'inover' in its name and the CPy source code doesn't reference it at all...

martin2250, 2021-10-31:
Ok, so if I configure other pins as input, they are also registered by the PIO:

`in_pin_count = 3` and these lines:
```
pins = [
  DigitalInOut(board.GP3),
  DigitalInOut(board.GP4),
  DigitalInOut(board.GP5),
  DigitalInOut(board.GP6),
  DigitalInOut(board.GP7),
]
```

produce the same results as `in_pin_count = 8`

urish, 2021-10-31:
> (I only used CPy because micropython isn't offered by wokwi.com

[image]

I guess this means we should probably be more explicit about the fact it's MicroPython on Raspberry Pi Pico?

> Ok, so if I configure other pins as input, they are also registered by the PIO:

So I guess it boils down to the IE (input enable) pin in the GPIO0...GPIO39 registers:

[image]

Right now we simply ignore it. But perhaps changing

https://github.com/wokwi/rp2040js/blob/235119548d16efe4818ed5122a5574dc0a6f5894/src/gpio-pin.ts#L141

to read

```ts
    return applyOverride(this.rawInputValue && this.inputEnable, this.inputOverride);
```

will do the trick?

martin2250, 2021-11-01:
> I guess this means we should probably be more explicit about the fact it's MicroPython on Raspberry Pi Pico?
Yes, I remembered using µP on your website and was wondering where that option went. Also it would be great to have the option to simulate C-SDK projects (but that is a matter for a separate issue).

> So I guess it boils down to the IE (input enable) pin in the GPIO0...GPIO39 registers:

Seems like it, IE is disabled here:
https://github.com/adafruit/circuitpython/blob/main/ports/raspberrypi/common-hal/microcontroller/Pin.c#L63
And it is enabled again here:
https://github.com/adafruit/circuitpython/blob/main/ports/raspberrypi/common-hal/rp2pio/StateMachine.c#L254
which calls the C-SDK:
https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/hardware_gpio/gpio.c#L34

urish, 2021-11-01:
Alright, I'll try to implement IE and let's see if it does the trick

urish, 2021-11-01:
Alright, the fix is up on wowki.com! Mind checking again now?

martin2250, 2021-11-01:
Tested with the code from above (https://wokwi.com/arduino/projects/313883083894948417), it behaves like in hardware. Thanks!

closed 2021-11-01 by martin2250

urish, 2021-11-01:
Amazing, thanks again for reporting and all the help figuring this out :)

martin2250, 2021-11-01:
sure thing, thank you for putting your time into this great simulation environment!