wokwi/rp2040js #86: PIO `jmp pin` branch is always taken

martin2250, 2021-11-01:
Hi again,

I found another potential issue with the PIO implementation. It seems like when using `jmp pin, xxx`, it always branches to the jump target. If I understand the documentation correctly, this branch should only be taken if the pin reads high.

Here is an example: https://wokwi.com/arduino/projects/314054657341129280

Start the simulation, set a breakpoint on PIO instruction 31 and play around with the switch.

Not sure if I configured something wrong, the [relevant line of code](https://github.com/wokwi/rp2040js/blob/master/src/peripherals/pio.ts#L252) seems fine.

urish, 2021-11-01:
Ok, I think I found the reason. I uploaded a fix, and got it to work with pin 2:

https://wokwi.com/arduino/projects/314060101811962434

Note that I also had to set the pin to input mode (otherwise the input is not enabled, apparently).

Not sure why it doesn't work with pin 0. My bet is that it's conflicting with the UART.

martin2250, 2021-11-01:
Works now, thanks for the quick fix!

That's good to know, didn't realize the REPL was injected by UART, I thought you had hooks in the USB stack. If this affects Pin 0/1 in general, maybe you should add a warning to the hover tooltip on those pins.

Cheers!

closed 2021-11-01 by martin2250

urish, 2021-11-01:
Hooray! 

The REPL indeed goes through USB, but there might be some remainders from the UART REPL in the code. I might also be wrong - need to check and see what's going on with pin 0