wokwi/rp2040js #93: Pico fails silently when initialising SD Card

craigdrayton, 2022-02-02:
I have a Wokwi project using micropython and the Pi Pico, with the wokwi-microsd-card module.

Upon attempting to initialize the SD card using the [standard micropython sdcard library](https://github.com/micropython/micropython/blob/master/drivers/sdcard/sdcard.py), the application hangs on Line 77 of the library `self.spi.write(b"\xff")`

Is this simply because SPI hasn't been implemented yet on Pico, or is something else going on? https://docs.wokwi.com/parts/wokwi-pi-pico

Cheers,
Craig

urish, 2022-02-02:
Hi Craig, SPI has been implemented (3c7c5c73).

Mind sharing the link to the your project so I can have a look?

craigdrayton, 2022-02-02:
Thanks for such a fast reply. Sure thing: https://wokwi.com/arduino/projects/322461882944324178

urish, 2022-02-02:
Thanks!

Apparently, the SIO wasn't connected on Wokwi's end (that's why the docs said it was unimplemented). 

I added the missing bits, and now your code seems to communicate with the SD Card. It seems to fail on this command:

```python
        if self.cmd(16, 512, 0) != 0:
           raise OSError("can't set 512 block size")
```

Command 16 hasn't been implemented yet, but the block size is 512 by default (at least in the simulator), so I believe you can remove this check. I haven't had a chance to look where it gets stuck next.

However, adding a [Logic Analyzer](https://docs.wokwi.com/guides/logic-analyzer) can be very valuable in diagnosing the communication. Here is a screen shot showing how pulse view decodes the beginning of the communication:

[image]

craigdrayton, 2022-02-03:
Thanks! The SD card seems to initialize correctly now, I can read sd.sectors and get 8MB.

Now I may be running up against limitations of the current wokwi sdcard module.
- I'm unable to mount the SD card (fails silently) `uos.mount(sd, "/sd")`
- If I skip mounting the SD card, `uos.listdir("/")` shows my project files
- I can then write to "/test01.txt" and then read the data, but the file does not persist

urish, 2022-02-03:
> I'm unable to mount the SD card (fails silently)

That's interesting. Any idea why?

> If I skip mounting the SD card, uos.listdir("/") shows my project files

That makes sense - Wokwi copies the project files to the Pi Pico's flash memory, so that's why you see them

> I can then write to, say, "/test01.txt" and then later read the data, but the file does not persist

This is true for both the internal flash memory and the SD Card - we don't persist the data at the moment. Still need to figure out what's the best way to go about it.

p.s. - I see your code also uses ulab. In general, it shouldn't be too hard to include it in Wokwi - we'd just need to have the uf2 here: https://github.com/wokwi/firmware-assets/tree/gh-pages/micropython, and then it'll be able to load it into Wokwi (there's already on with ulab for the ESP32)

urish, 2022-02-03:
Thanks to @v923z, we now have a ulab-enabled firmware for the Raspberry Pi Pico as well!

https://wokwi.com/arduino/projects/322595929479709267

If you want to switch to it in your project, change the value of the "env" attr in your diagram.json to "micropython-ulab-26faf74d5-ccd9481"

craigdrayton, 2022-03-03:
Thanks @urish I really appreciate the quick responses (apologies for my delay on this one).

closed 2022-03-03 by craigdrayton