wokwi/rp2040js #98: ILI9341 freeze with adafruit_ili9341 lib

danilinao, 2022-06-09:
When i'm try run this code:
```
import board
import busio
import displayio
import terminalio
import adafruit_ili9341
from adafruit_display_text import label

dc=board.GP5
rst=board.GP6
blk=board.GP7
cs=board.GP8
 
spi = busio.SPI(clock=board.GP2, MOSI=board.GP3, MISO=board.GP4)
 
displayio.release_displays()
display_bus = displayio.FourWire(spi, command = dc, chip_select=cs, reset=rst)
 
display = adafruit_ili9341.ILI9341(display_bus, width=320, height=240, rotation = 180)

splash = displayio.Group()
display.show(splash)
```
the emulator freezes and the following messages appear in the js console

[image]

danilinao, 2022-06-09:
Sure, https://wokwi.com/projects/333969171970785876

urish, 2022-07-20:
Just updating that I managed to reproduce the issue, but still haven't had a chance to dig into it. It's on my radar, though.

guysv, 2023-09-02:
reporting here beacuse it seems related:

I tried to run some micropython SPI code on the emulator, and found out that [spi.write](https://docs.micropython.org/en/latest/library/machine.SPI.html#machine.SPI.write) fails to write any any buffer of length longer than 32 (included). no data ever reaches the rp2040js's spi data callback, not even the first 32 bytes.

e.g given the firmware:
```python
from machine import SPI
spi = SPI(0)
spi.write(b"0123456789abcdef0123456789abcde")
spi.write(b"0123456789abcdef0123456789abcdef")
```
the [rp2040js spi callback](https://github.com/wokwi/rp2040js/blob/master/src/peripherals/spi.ts#L66) will only be called for every byte in the first write.

investigating the micropython repo, I figured [ports/rp2/machine_spi.c#machine_spi_transfer](https://github.com/micropython/micropython/blob/c0d4c604e6a140c0f2967e1b43fd94d0b029c73f/ports/rp2/machine_spi.c#L228-L287) is the function responsible for actually calling the pico sdk [spi_write_blocking](https://www.raspberrypi.com/documentation/pico-sdk/hardware.html#rpip1c71a31c85168031ca89) func. Look at its code: if the input buffer length is >= 32, then data is sent to SPI via DMA which in rp2040js **is not yet implement**

good news are DMA already exists in rp2040js, it just doesn't seem wired into the SPI peripherial. Im looking into the fix 🙏

PR #131 (2023-09-22): Support SPI DMA
potentially closes #98, see comment there.

must admit this looks even slower than straight spi register to callback path... we probably want to squeeze the CPU better here.
closed 2023-09-22 by urish