wokwi/rp2040js #99: How to simulate esp32?

BFjacky, 2022-06-17:
Wokwi is a great website. It is unbelievable to simulate many microcontroller unit.It looks like some kind of magic.I also try to build a website that can simulate "micropython on esp32" just like user can create at 'https://wokwi.com/'. I want to simulate esp32 instead of Raspberry Pi Pico. But i dont know how to achieve it.Does this repository can help me implement that? .Is 'https://wokwi.com/' an open source project? Looking forward to your answer. Thankyou very much in advance!

urish, 2022-06-17:
Thanks! Some parts of Wokwi are open, some are not. The ESP32 simulator is not currently open. It might be in the future. 

I'm curious to know, for what purpose would you like to build the "micropython on ESP32" website?

BFjacky, 2022-06-17:
Thanks for your reply! I'm conceiving my graduation project and trying to make a simulation application on browser. Honestly, I'm also really trying to figure out what happened after I upload the micropython code to the esp32.

urish, 2022-06-17:
In that case, I think the best bet would be either to use the Web Assembly build of MicroPython, use MicroPython with the RP2040 (using rp2040js), or using their [Unicorn build](https://micropython.org/unicorn/).

Simulating the ESP32 is a huge undertaking. For reference, you can watch [behind the scenes for RP2040js implementation](https://www.youtube.com/watch?v=POCUbWhBXVA&list=PLLomdjsHtJTxT-vdJHwa3z62dFXZnzYBm&index=1), which took over 70 hours of hard work to implement.

BFjacky, 2022-06-17:
Wow..That's really a complex project.I think i should start with the series of lessons that you recommend.I don't know much about the difference between Esp32 and Raspberry Pi. In my mind, the main difference may be the execute instruction.

closed 2022-06-17 by BFjacky

urish, 2022-06-18:
If you are just getting started with the Pi, then the following 5-hour course can help you to wrap your head about the internals:

https://www.youtube.com/watch?v=OLV-TSRTTE8&list=PL_tws4AXg7auiZHZsL-qfrXoMiUONBB0U&index=1

As far as the differences between the Pi Pico and the ESP32, there are many :-) They have a different architecture (ARM vs Xtensa), thus a different instruction set, the ESP32 has many more peripherals (WiFi, Bluetooth, Ethernet are some of the more notable ones), and a low power co-processor. The Pi Pico, on the other hand, has a unique programmable I/O peripheral (PIO) and USB support.