wokwi/rp2040js #101: CDC Issues with MicroPython 1.19.1 on 0.17.0

jmdevy, 2022-07-19:
Hi,

I am trying to use MicroPython Pico 1.19.1 in the demo program put the CDC serial output is a bit garbled. I am using 0.17.0 of this emulator.

**Expected (Pico MicroPython 1.17 + 0.17.0 emulator):**
```
>>> help()
Welcome to MicroPython!

For online help please visit https://micropython.org/help/.

For access to the hardware use the 'machine' module.  RP2 specific commands
are in the 'rp2' module.

Quick overview of some objects:
  machine.Pin(pin) -- get a pin, eg machine.Pin(0)
  machine.Pin(pin, m, [p]) -- get a pin and configure it for IO mode m, pull mode p
    methods: init(..), value([v]), high(), low(), irq(handler)
  machine.ADC(pin) -- make an analog object from a pin
    methods: read_u16()
  machine.PWM(pin) -- make a PWM object from a pin
    methods: deinit(), freq([f]), duty_u16([d]), duty_ns([d])
  machine.I2C(id) -- create an I2C object (id=0,1)
    methods: readfrom(addr, buf, stop=True), writeto(addr, buf, stop=True)
             readfrom_mem(addr, memaddr, arg), writeto_mem(addr, memaddr, arg)
  machine.SPI(id, baudrate=1000000) -- create an SPI object (id=0,1)
    methods: read(nbytes, write=0x00), write(buf), write_readinto(wr_buf, rd_buf)
  machine.Timer(freq, callback) -- create a software timer object
    eg: machine.Timer(freq=1, callback=lambda t:print(t))

Pins are numbered 0-29, and 26-29 have ADC capabilities
Pin IO modes are: Pin.IN, Pin.OUT, Pin.ALT
Pin pull modes are: Pin.PULL_UP, Pin.PULL_DOWN

Useful control commands:
  CTRL-C -- interrupt a running program
  CTRL-D -- on a blank line, do a soft reset of the board
  CTRL-E -- on a blank line, enter paste mode

For further help on a specific object, type help(obj)
For a list of available modules, type help('modules')
>>>
```

**Actual (Pico MicroPython 1.19.1 + 0.17.0 emulator):**
```
>>> help()
Welcome to MicroPython!

For online help please visit https://chine' module.  RP2 specific commands
  machine.Pin(pin) -- get a pin, eg machine.Pin(0)
  machine.Pin(pin, m, [p]) -- get a pin and configure it for I, low(), irq(handler)
  machine.ADC(pin) -- make an analog obje  machine.PWM(pin) -- make a PWM object from a pin
             readfrom_mem(addr, memaddr, arg), writeto_mem(addr,
  machine.SPI(id, baudrate=1000000) -- create an SPI object (i_readinto(wr_buf, rd_buf)
  machine.Timer(freq, callback) -- cr    eg: machine.Timer(freq=1, callback=lambda t:print(t))

Pins are numbered 0-29, and 26-29 have ADC capabilities
Pin.PULL_UP, Pin.PULL_DOWN

Useful control commands:
  CTRL-C --  CTRL-D -- on a blank line, do a soft reset of the board
  CTRL-E -- on a blank line, enter paste mode

For further he modules, type help('modules')
>>>
```

urish, 2022-07-20:
Thanks for reporting! 

Seems like the new MicroPython version uses USB double buffering (section 4.1.2.5.3 in the datasheet). We don't implement that yet.

urish, 2022-07-20:
Using `git bisect`, I found that the problem started with micropython/micropython@587339022689187a1acbccc1d0e2425a67385ff7

jmdevy, 2022-07-20:
I used `git revert 5873390`, rebuilt MicroPython, and now serial is working as it should again.

PR #107 (2022-11-24): Fix #101
Fix #101 

MicroPython Pico 1.19.1 `help()` output as expected by implement double buffer for USB.
commit 641393d5 (2022-11-24): feat(usb): implement double buffering #107

fix #101
closed 2022-11-24 by urish