wokwi/rp2040js #114: Allow specific image file.

BrianPugh, 2023-01-24:
Would you be interested in having the micropython image be specifiable via CLI args for the `start:micropython` script? I'm not a javascript/typescript dev, but if you're open to the idea, I think I could try and make a PR. Seems like something like `minimist` is the way to go for argument parsing?

urish, 2023-01-25:
Sounds like a useful option. What is your specific use case for this? What are you working on?

BrianPugh, 2023-01-25:
I'm using rp2040js as an integration tester for [Belay](https://github.com/BrianPugh/belay). I would like to swap between 2 images (micropython and circuitpython), so having the image be a CLI option seems like a natural choice.

urish, 2023-01-30:
Cool use case, thanks for explaining!

mingpepe, 2023-01-30:
This feature also useful for me.

BrianPugh, 2023-01-30:
resolved in #116

closed 2023-01-30 by BrianPugh