wokwi/rp2040js #122: RPi Pico Error `WARN: starting new transfer on already active ep 0 out`

aaronjamt, 2023-02-26:


urish, 2023-02-26:
Got it! Unfortunately, I don't have the bandwidth to look into this at the moment. 

In case someone else wants to look into this, it can be reproduced with rp2040js by downloading the snowflake.zip attachment from the first comment in this issue, adding the following code at the end of `demo/micropython-run.ts`:

```ts
mcu.uart[0].onByte = (value) => {
  process.stdout.write(new Uint8Array([value]));
};
```

and then running the simulator it as follows:

```
npm run start:micropython -- --gdb --image snowflake/build/snowflake.uf2
```

aaronjamt, 2023-02-26:
> Got it! Unfortunately, I don't have the bandwidth to look into this at the moment.
>
> In case someone else wants to look into this, it can be reproduced with rp2040js by downloading the snowflake.zip [...]

No worries if you don't have time/bandwidth right now, it's more of an annoying warning than anything actually problematic. As it turns out, the corrupt serial output was my bad because I was using GPIOs 0 and 1 without realizing that the simulated UART connection was _also_ using those pins (so any time my code changed the GPIO state it messed with the UART data).

Just one quick question: your commands to reproduce say `npm run start:micropython` but I'm using the RPi Pico C/C++ SDK (not MicroPython). Could that have something to do with it?

urish, 2023-02-26:
Thanks for the update!

Regarding your question, not at all - the micropython demo app is called that way because we used it initially to test micropython, but it can run any Pi Pico UF2 binary. It does have some defaults that work well for micropython (such as using Serial over USB and not the standard UART). But you can use it to run any binary built with he RPi Pico C/C++ SDK.

closed 2023-12-23 by aaronjamt