wokwi/rp2040js #123: rfc2217ServerPort issue

realJoshByrnes, 2023-03-08:
Not sure where to put this as there doesn't seem to be a repo for the Wokwi VSCode Extension, but it relates the the rp2040js as that's the component I'm having issues with.

### wokwi.toml

```toml
[wokwi]
version = 1
firmware = './firmware/kaluma-rp2-pico-1.1.0-beta.1.uf2'
elf = ''
rfc2217ServerPort = 4000
```

### diagram.json

```json
{
  "version": 1,
  "author": "Josh Byrnes",
  "editor": "wokwi",
  "parts": [
    {
      "type": "wokwi-pi-pico",
      "id": "pico",
      "top": 0,
      "left": 0,
      "attrs": { "env": "kaluma-1.0.0" }
    }
  ]
}
```

## Issue

Current setup involves Wokwi Simular (VSCode) containing only a rPi Pico, and Kaluma (JS) firmware.

Whilst a connection can be established and REPL works as expected, it appears that some binary data is lost/encoded, causing a `kaluma flash` to never end. Everything else works as expected.

Notes:
* Kaluma works fine with real hardware via the COM port
* I have tried with multiple RFC2217 software such as [Perle's Trueport](https://www.perle.com/downloads/trueport.shtml) (recommended) and [Electronic Team's Serial to Ethernet Converter](https://www.eltima.com/products/serial-over-ethernet/) (commercial)

urish, 2023-03-08:
Thanks for reporting!

Can you try removing the `"env": "kaluma-1.0.0"` attr from the diagram?

The "env" might confuse Wokwi and it will end up doing the wrong thing

realJoshByrnes, 2023-03-08:
> Can you try removing the `"env": "kaluma-1.0.0"` attr from the diagram?

Just tried and it results in the same issue. I can only think it's directly a RFC2217 encoding issue (are you decoding UTF8 on an ASCII protocol?). Unfortunately, I couldn't find the source to investigate further.

P.S. The console looks so much nicer with env set to kaluma-1.0.0!

urish, 2023-03-08:
RFC2217 encoding issue is not very likely, since we got esptool to work over RFC2217 and flash a simulated ESP chip successfully. Can you share your entire setup in a GitHub repo so I can have a look?

Nicer console - are you referring to the terminal? You can achieve this by setting "display" to "terminal": https://docs.wokwi.com/guides/serial-monitor#display

realJoshByrnes, 2023-03-09:
OS is Windows 11
* VSCode with the Wokwi extension and the above config files (listens on TCP port 4000)
* [Perle Trueport](https://www.perle.com/downloads/trueport.shtml) installed, with a client forwarding COM3 to TCP port 4000
* [Kaluma CLI](https://github.com/kaluma-project/kaluma-cli) installed using `npm install -g @kaluma/cli` (requires nodejs)

Firmware can be downloaded from https://kalumajs.org/download/

A simple index.js could just contain
```js
console.log('Hello world');
```

Now, from the command line, we should be able to run the following
```cmd
$ kaluma flash ./index.js --bundle --shell --COM3
```