wokwi/rp2040js #127: How to simulate DHT22 and use it with RP2040 to read temperature and humidity?

TBWXY777, 2023-04-19:
Hi, I am trying to use the DHT22 temperature and humidity sensor with RP2040, and I want to simulate the sensor.
I am hoping to find some guidance on how to simulate the DHT22 sensor and use it with RP2040 to read temperature and humidity. I have attached my current Arduino code below:
```
#include "DHT.h"

#define GPIOPIN 2
#define DHTTYPE DHT22

DHT dht(GPIOPIN, DHTTYPE);

void setup() {
  Serial1.begin(9600);
  dht.begin();
}

void loop() {
  delay(200);
  float h = dht.readHumidity();
  float t = dht.readTemperature();
  Serial1.print("Humidity: ");
  Serial1.print(h);
  Serial1.print("%  Temperature: ");
  Serial1.print(t);
  Serial1.println("°C");
}
```
Thank you very much for your help!

urish, 2023-04-20:
Hello, nice to meet you!

I'm curious, what are you building?

Re DHT22 - you'll have to implement the protocol. Since the protocol is timing sensitive, I suggest to wait a bit until I have a chance to push the clock changes for wokwi/wokwi-features#186 (see [my comment here](https://github.com/wokwi/rp2040js/pull/117#issuecomment-1496636555)). 

Another way to go about this would be to read the symbols from the ELF file, intercept the calls to `readHumidity()` and `readTemperature()`, and return a fake value instead of calling the actual code. You could do it by replacing the [blTaken()](https://github.com/wokwi/rp2040js/blob/ffd540a975134b38a9cf4244ba032175fef3c846/src/cortex-m0-core.ts#L92) hook with your own implementation that would compare the `PC` value with the symbol you read from the ELF file, and in case of a match will put the return value in the relevant register (r0, if I'm not wrong), and update PC with the value of LR to immediately return from the call.

TBWXY777, 2023-04-21:
> Hello, nice to meet you!
>
> I'm curious, what are you building?
>
> Re DHT22 - you'll have to implement the protocol. Since the protocol is timing sensitive, I suggest to wait a bit until I have a chance [...]

I am glad to receive your reply. I am not building anything, I am just curious about the components of wokwi-elements. Recently, I successfully simulated the interaction between LCD1602 and PICO of wokwi-elements, which I found very interesting. Therefore, I want to try simulating DHT22. However, simulating DHT22 is a bit difficult for a beginner like me. Your suggestions have given me a lot of inspiration. Thank you very much.

urish, 2023-04-23:
Good luck!

closed 2023-04-23 by urish