wokwi/rp2040js #128: Is there a BUG in the RX function of UART in Pico?

TBWXY777, 2023-05-25:
Hello urish, I used the RX function of UART on the wokwi platform with both Uno and Pico. Uno works normally, but when I input to the serial port on Pico, there is no corresponding output and the browser becomes very stuck. Every time I input to the serial port, the stuckness will increase once.

    The two projects tested on the wokwi platform are the following two.
    Pico:https://wokwi.com/projects/365624376504121345
    Uno:https://wokwi.com/projects/365624339908814849

    Maybe there is a problem with the way I use it? Could you please tell me the correct way to use it and provide an example of the RX function of UART in Pico if possible?
    Looking forward to your reply~~

urish, 2023-05-25:
Hi, thanks for reporting!

There seems to be a bug in your Pi Pico code - it should call `Serial1.read()`, not `Serial.read()`. 

Nevertheless, it doesn't work even with `Serial1.read()`, seemingly keeping in the `while` loop forever.

Have you tried this code on the actual Pi Pico?

urish, 2023-05-25:
Switching to `Serial1` and compiling with the [arduino community core](https://docs.wokwi.com/parts/wokwi-pi-pico#arduino-core) fixes the issue:

https://wokwi.com/projects/365686547754485761

TBWXY777, 2023-05-25:
Thank you for your reply, it is very helpful. I would like to know if I don’t use Pico on the platform, but use the code in RP2040js, similar to the following.
```
rp2040.loadBootrom(bootromB1);
rp2040.logger = new ConsoleLogger(LogLevel.Error);
loadHex(hex, rp2040.flash, 0x10000000);
rp2040.core.PC = 0x10000000;
rp2040.execute();
```
Then how should I switch between the official version and the community version?

urish, 2023-05-25:
You just compile the code with the following Arduino core: https://github.com/earlephilhower/arduino-pico

I'm curious - what project are you working on?

TBWXY777, 2023-05-25:
Wow, you reply so fast. I am trying to interact between multiple virtual devices because it seems that I can’t use multiple Picos or multiple Unos on wokwi at the same time. I have used the Arduino compilation library before to get the official and community version of the Hex code, but it seems that when the official version of Hex is injected into Pico, there will be some problems, such as the delay of `delay()` is not correct. Do I need to configure something else in the code?

urish, 2023-05-25:
Trying :-)

Well, we're working on support for simulating multiple virtual devices (wokwi/wokwi-features#186). 

As part of that effort, I'm planning to rewrite the clock implementation of this library, which will make it easier to implement accurate timing (e.g. for `delay()`). Right now, the timing is a bit flakey as the clock is tied to JavaScript timers.

TBWXY777, 2023-05-26:
Okay, looking forward to your update~~

closed 2023-05-26 by TBWXY777