wokwi/rp2040js #135: SSD1306 broken on micropython 1.20+

treysis, 2023-11-16:
Trying to interface the board-ssd1306 component via I2C gives the following error:

[image]

Setting micropython environment in `diagram.json` to below 1.20 works, e.g. `micropython-20220618-v1.19.1`. But since 1.20 it's broken (same problem with 1.21).

urish, 2023-11-16:
Thanks for reporting! Can you please share a link to a demo project that demonstrates the issue?

treysis, 2023-11-16:
Sure, see this for example:
https://wokwi.com/projects/381549021386487809

urish, 2023-11-16:
Thanks, issue reproduced. Working on a solution.

treysis, 2023-11-16:
For reference, here is a (generic?) ESP32 demo which works with 1.20+:
https://wokwi.com/projects/305568836183130690

urish, 2023-11-16:
Thanks! Fix is up, can you please check?

treysis, 2023-11-16:
Wow, that was lightning fast! Yes, it's working now. 👍 Thanks a lot!

closed 2023-11-16 by treysis