wokwi/rp2040js #141: Does the wokwi simulator support the round-robin method of reading multiple ADC channels in pico?

rmoh12, 2024-05-28:
Hi, I tried to change the dma_capture pico sdk example code to read multiple ADC inputs in a round-robin method, but only ADC0(pin26) is read. All my efforts to implement this method in wokwi simulator ended with the same result. at the address
https://forums.raspberrypi.com/viewtopic.php?t=350378
 I saw that this work was done. So I tried to run the final code in the wokwi simulator, but the result was the same, that is, only ADC0 (pin26) is read.
https://wokwi.com/projects/399099758831776769
It seems that the wokwi simulator does not support the round robin method for pico. Please guide me if anyone has any information on this matter. Thankful

urish, 2024-05-28:
Hi! The ADC does seem to implement multi-shot read in a round robin fashion, but I'm not sure how well tested this specific code path is. If you are interested, you can take a look at the implementation here:

https://github.com/wokwi/rp2040js/blob/7243fc7a3e17b6a80b4899627bcc3e7cfe594f74/src/peripherals/adc.ts#L214-L234

rmoh12, 2024-05-29:
hi, thank you for your reply.
I did not work much in the field of java script. I looked at the part of the code you gave. It doesn't seem like there is a problem with that part. But it is important what value is passed to the writeUint32 function in the value and offset parameters. As a result, the behavior of the program should be checked at the time of execution. I don't know what the main code of the program is or how to run it. For the sake of it, I decided to test my code in the real environment. It really worked. But not in the wokwi simulator. So we must have a problem here and the wokwi simulator does not work properly in the round robin method.  I really don't know how to fix it.
I remind you that I am not very fluent in Java, but if you can guide me on how to run this simulator in the vs code environment, I might be able to help.
If not, I think someone should bother to fix it.