wokwi/rp2040js #142: RP2350 support

c1570, 2025-01-31:
Just as a pointer for anyone interested:
https://github.com/c1570/rp2040js features some RP2350 support (RISC-V only).

urish, 2025-01-31:
Great! What's the end goal?

c1570, 2025-02-09:
I'm mostly using it for https://github.com/c1570/Connomore64 and it's running that already so not sure.

I'll probably implement interrupts (because most normal projects need those) and GPIO/PIO changes (those additional GPIOs are interesting in the context of other retro tinkering). Supporting MicroPython would also be nice.

Of course it would also be interesting to get this into Wokwi but that's probably a long way to go...

c1570, 2025-02-12:
Difficult to say, around 50 hours probably, including looking at RISC-V for the first time.

Interrupts are there now but need some debugging. Xh3irq is a nifty feature but a bit complex.

Also, I merged most of the recent rp2040js changes apart from reformatting/ESM/tsx changes.
For rp2350js, tsx is _much_ slower than ts-node for some reason (tsx runs is at only about 30% of ts-node's speed even though initial startup is much faster with tsx).

urish, 2025-02-16:
That's surprising, at the end of the day, both are just removing the types. IIRC, in recent node versions (22?) you can just use  --experimental-strip-types to directly run TypeScript code, without any extra "helpers"

ohault, 2025-03-23:
What is the current status about Pico2 support (including Arm) ?

urish, 2025-03-23:
I'm not actively working on that. If anyone is up to sponsoring the development I'm happy to discuss.