wokwi/rp2040js #144: PWM peripheral hangs occasionally

c1570, 2025-02-18:
It seems the PWM peripheral can be brought into a state in that it reschedules an alarm immediately in an infinite loop.
See https://github.com/c1570/rp2040js/commit/0d8e5b46e2db12076d7f2faabfb5c80814cda7fd for a test case.

This PWM configuration is used by, e.g., PicoDVI (though looking at that code https://github.com/Wren6991/PicoDVI/blob/8ecccce7b7b7ac8a0f045974efb97852e78d2306/software/libdvi/dvi_serialiser.c#L44 I'm not sure how exactly those register settings can happen ;) ).

c1570, 2025-02-20:
It seems that Timer32PeriodicAlarm.handleAlarm() does the callback then calls schedule() which enqueues the next alarm immediately (since targetValue = rawCounter).

https://github.com/c1570/rp2040js/commit/18008b43e557b1d6128a216074d327e7d5f4a04d fixes this but not sure if that's a clean/correct solution.

Also, https://github.com/c1570/rp2040js/commit/dc1b06da0023052cb763b6c24b8379f71b8fb9b3 adds some strings to timer objects so that this kind of behaviour is easier to debug.

urish, 2025-02-22:
Thanks for reporting, pushed your fixed. Released at 1.0.2.

I'd also love to include the test case, but as it's currently written, it doesn't have any expectation that fails with the wrong code. If you can rewrite it to actually fail with the previous code, and pass with the new code, I'd love to add it.

commit e258a5ae (2025-02-22): fix(pwm): peripheral hangs occasionally #144

close #144
closed 2025-02-22 by urish