wokwi/rp2040js #147: GPIO listener not firing on SPI enabled pins.

jorbo, 2025-06-16:
I am trying to connect the SPI pins to another simulated device. I've added a listener to the specific GPIO pins that are associated with the SPI setup in the firmware, but the listeners are never called. 

Here is the firmware and the way I am adding the listener.

```c++
#include <stdio.h>
#include "pico/stdlib.h"
#include "pico/binary_info.h"
#include "hardware/spi.h"
#include <stdlib.h>
#include "time.h"
#include "stdio.h"
#include <boards/pico.h>

#define SPI_PORT spi0
#define PIN_MISO PICO_DEFAULT_SPI_RX_PIN
#define PIN_CS   PICO_DEFAULT_SPI_CSN_PIN
#define PIN_SCK  PICO_DEFAULT_SPI_SCK_PIN
#define PIN_MOSI PICO_DEFAULT_SPI_TX_PIN

#define BUF_LEN         1
int main()
{
    stdio_init_all();
    srand(time(0));

    spi_init(SPI_PORT, 1000*1000);
    gpio_set_function(PIN_MISO, GPIO_FUNC_SPI);
    gpio_set_function(PIN_SCK, GPIO_FUNC_SPI);
    gpio_set_function(PIN_MOSI, GPIO_FUNC_SPI);
    gpio_set_function(PIN_CS, GPIO_FUNC_SPI);

    const uint LED_PIN = PICO_DEFAULT_LED_PIN;
    gpio_init(LED_PIN);
    gpio_set_dir(LED_PIN, GPIO_OUT);

    uint8_t out_buf[BUF_LEN], in_buf[BUF_LEN];   
    
    for (size_t i = 0; ;++i){
        out_buf[0] = rand() % 255;
        gpio_put(LED_PIN, 1);
        spi_write_blocking(spi_default, out_buf, BUF_LEN);
        gpio_put(LED_PIN, 0);
        busy_wait_ms(500);
    }
}

```

```javascript
  mcu.gpio[19].addListener((state: GPIOPinState, oldState: GPIOPinState) => {
    console.log('GPIO 19 state changed:', state);
  });

```

urish, 2025-06-16:
Looks like a duplicate of #105 ?

jorbo, 2025-06-19:
Seems like we're asking similar questions. To put it more generally, how do we mimic the behavior of the SPI pins with the current SPI implementation. I see the discusion about the CS pin in #105, I think a similar approach can be used for the RX/TX pins but I am not sure about the SCK pin.

urish, 2025-06-21:
The short answer is that you have to implement the actual SPI pin logic yourself - rp2040js only gives you the high level interface, it doesn't actually implement the low level protocol (the same is also true for UART / I2C / USB).